PhotonVision/photonvision #88: 640x480 calibration will always fail

mcm001, 2020-08-14:


mcm001, 2020-08-17:
Except that it doesnt

Bankst, 2020-09-16:
Reproduced at 320x240 also - @XzibitGG reported it may have something to do with using Point3 to store points.

XzibitGG, 2020-09-16:
Yeah, object points should contains points in floating point integers instead of doubles, but there does not appear to be a Point3f method instead of Point3 to use. Even then, I think the next step is to see if unit tests pass with equivalent settings and snapshots used to calibrate from the front end.

mcm001, 2020-12-16:
Even if it doesn't fail, extrinsics often have huge numbers

gerth2, 2021-11-22:
Is this one still a thing? I've run a number of cal's recently without issue. I hate the 'code works, don't know why' closure issue, but curious if anyone's replicated it recently.

closed 2024-12-21 by spacey-sooty