PhotonVision/photonvision #124: LED Dimming Doesn't Work on Gloworm

carlosgj, 2020-09-16:
In the latest dev build (photonvision-dev-dev-13-g66a5fa8.jar), the Brightness slider in the Settings page doesn't dim a Gloworm's LEDs properly. If the slider is set to 100, the LEDs are fully on. If the slider is set to any other value, the LEDs are fully off.

mcm001, 2020-09-16:
Thanks for noticing. This is to do with how we are (or aren't) handling PWM frequency ranges in the backend, I think. I'll look more into it.

carlosgj, 2020-09-16:
FWIW, for "ledPWMFrequency", I've tried 30000 (per the Gloworm image) and 800 (per `photon-server/src/test/resources/hardware/HardwareConfig.json`) with no noticeable differences.

mcm001, 2020-09-16:
This seems to be related to an integer division bug in PiGPIO.

Bankst, 2020-09-17:
Fixed  in #127, awaiting further testing to confirm

pietroglyph, 2020-09-17:
Confirmed working on my Gloworm too.

closed 2020-11-08 by pietroglyph