PhotonVision/photonvision #125: Camera Exposure Control Doesn't Work on Gloworm

carlosgj, 2020-09-16:
The camera shutter speed seems to be 0 (or very small) when the exposure slider is set to 0, and some fixed value when the slider is greater than 1. 

The shutter should be able to be set from 1 to 6,000,000 microseconds (or some subset of that range).

pietroglyph, 2020-09-16:
This is a known issue, and there are fixes in V4l2/CSCore to make this work correctly, but the camera backend on the Raspberry Pi 3 is going to be completely replaced with one that uses MMAL and does GPU acceleration (and incidentally does exposure control correctly) so we're going to wait to fix this until the GPU stuff gets merged.

gerth2, 2020-09-21:
FYI: Confirmed something nearly identical with a Pi V2 camera and a Pi 4. 10-4 on waiting for new core to support it.

I do like the idea of setting/displaying it in terms of physical units - drives education at the UI level about what the underlying thing the user is actually tweaking.

Bankst, 2020-12-06:
Fixed in #146

closed 2020-12-06 by Bankst