PhotonVision/photonvision #210: Image save rotation/position sometimes wrong

mcm001, 2021-01-01:


mcm001, 2021-01-01:
[image]
[image]

mcm001, 2021-01-01:
I can't reproduce that weirdness on the input images

samfreund, 2025-04-08:
Probably OBE, but I'll let Matt close it as I want to be sure

mcm001, 2025-04-08:
Unable to reproduce with a variety of divisors and rotations. Closing as OBE.

[image]
[image]
[image]
[image]
[image]
[image]

closed 2025-04-08 by mcm001