PhotonVision/photonvision #260: Add configuration in settings window for NT connection

gerth2, 2021-03-11:
Add some more configuration. Specifically, the use case to address is when photonvision and a simulated robot are both running on a desktop pc. 

Suggestion:
1) In client mode, allow the server URL and port to be configured.
2) in server mode, allow the port to be configured.

mcm001, 2021-10-02:
So basically just add a port number?

gerth2, 2021-10-02:
That's the main chunk.

There was also some question as to whether, in client mode, arbitrary servers could be supported (ie, not on the 10.TE.AM.** subnet, or localhost, or similar).

Could be maybe a switch for basic/advanced - basic just lets them set team number and assumes client, advanced allows server and arbitrary url/port ?

gerth2, 2021-11-22:
https://github.com/PhotonVision/photonvision/pull/308 should cover this, all hunky dory.

closed 2021-11-22 by gerth2

etiennebeaulac, 2023-01-18:
+1, it would be really useful to be able to hook PhotonVision to any NetworkTables server (such as a robot simulation), as said by @gerth2 here https://github.com/PhotonVision/photonvision/issues/260#issuecomment-932771181.