PhotonVision/photonvision #270: Photonlib simulation: hasTargets() is always false

mcm001, 2021-04-07:
https://discord.com/channels/725836368059826228/725848198794706994/829111851932712990

gerth2, 2021-07-29:
Per https://discord.com/channels/725836368059826228/725848198794706994/830076534076342272 - I think this is resolved?

More broadly, some graphical way to represent what is being seen in Sim could help users debug it in the future.

KyleQ1, 2021-08-05:
I agree a more graphical way to represent what's happening would be great. Did a lot of trial and error to get it working. I was able to get hasTargets() to work and just needed to do more testing.

gerth2, 2021-11-22:
@Aadit-Ambadkar any updates on this from your end? I have a few samples and test code showing `hasTargets()` can return true when expected? I've stopped investigation, but can re-open if you have a repo or example of it not working?

mdurrani808, 2023-10-06:
Closing, feel free to reopen as needed. Simulation architecture was rewritten and this issue is no longer relevant.

closed 2023-10-06 by mdurrani808