PhotonVision/photonvision #272: Give gain slider more flexibility

calcmogul, 2021-04-21:
There's currently one "gain" slider that ties the red and blue channels together, but low gain/exposure settings results in green images under fluorescent lighting. This is a problem if the target is lit green; the user has to use a much wider hue passband.

One solution to this would be separate red, green, and blue exposure sliders. A button could be added to autoselect the settings for them when the targeting LEDs are off (aka "auto white balance").

MATLAB's docs mention a few different algorithms that I imagine would be a starting point.
https://www.mathworks.com/help/images/comparison-of-auto-white-balance-algorithms.html

OpenCV has this, but idk if it's helpful: https://docs.opencv.org/4.5.2/d7/d71/classcv_1_1xphoto_1_1GrayworldWB.html

mcm001, 2021-04-21:
Per discussion on Discord:
"Gain" right now adjusts the red and blue balances together. They're scaled in [0, 8]. Green is fixed.
We can separate the red and blue balances, as well as expose contrast/saturation/iso. Those might best go under an advanced dropdown or something, as it's starting to get cluttered.

mcm001, 2021-10-02:
The gain slider was previously broken; latest version shouldn't have this issue. Do we still think it's valuable to split the gain slider?

gerth2, 2021-10-02:
After playing with it myself .. I vote no, single slider works plenty well.

closed 2021-11-24 by gerth2