PhotonVision/photonvision #274: PhotonVision not Working on Romi Robot

zagdrath, 2021-07-26:
I followed both [https://docs.photonvision.org/en/latest/docs/getting-started/installation/romi.html] and [https://docs.photonvision.org/en/latest/docs/getting-started/installation/coprocessor-image.html#other-debian-based-co-processor-installation] while installing PhotonVision to my romi robot and  when I do systemctl status  photonvision.service an error come sup and says this.
[image]

I am unable to access to PhotonVision web interface by going to http://10.0.0.2:5800/ which is what the guide says to go to.

zagdrath, 2021-07-27:
I do believe the issue is that the systemd service is not running as root even though it should by default.

mcm001, 2021-07-27:
Yeah that seems to line up. Photonvision needs to run as root on Linux usually yes.

Daltz333, 2021-07-27:
The solution here is that the system needs to be in RW mode.

```
systemctl stop photonvision.service
rw
systemctl start photonvision.service
```

The install script should detect if the host system is a romi (or have in docs) and disable RW (or other workaround) for the romi since photonvision handles this.

Daltz333, 2021-07-27:
More information: Romi system can be detected via existence of
```
pi@photonvision(rw):~$ ls /service/wpilibws-romi
log  run  supervise
```

gerth2, 2021-11-24:
Also update.... the latest development version (and eventual 1.9 release) should have a change which allows running without root. You won't be able to save settings, but it should at least execute.

mcm001, 2025-03-20:
Closing in favor of #1452

closed 2025-03-20 by mcm001