PhotonVision/photonvision #276: VMX-Pi CoProcessor CPP compatibility

MrRSquared, 2021-08-18:
Hello,
I am wondering about using photon on a separate Raspberry pi with the VMX-Pi as our robot (in FRC Library mode).  I did post over in CD, but received no response, and this inquiry is related to work we are doing this week. So, I am posting here too. Sorry for the double post.

We have the coprocessor working, and everything works with Java (the robot receives Photon Data. However, in CPP, our project will not build. we get the following error.

```
In file included from C:\...\.gradle\caches\transforms-2\files-2.1\7dbabd40d14cb830cf74c2fa32bc8b3d\PhotonLib-cpp-2021.1.2-headers/photonLib/PhotonCamera.h:27,
                 from C:\...\vmxSummer2021\beginning\src\main\include/Robot.h:16,
                 from C:\...\vmxSummer2021\beginning\src\main\cpp\Robot.cpp:8:
C:\...\.gradle\caches\transforms-2\files-2.1\7dbabd40d14cb830cf74c2fa32bc8b3d\PhotonLib-cpp-2021.1.2-headers/photonlib/PhotonPipelineResult.h:23:10: fatal error: units/time.h: No such file or directory
 #include <units/time.h>
          ^~~~~~~~~~~~~~
compilation terminated.
```

I am assuming that the issue is that the VMX-Pi uses WPILib 2020.3.2 instead of 2021. Does anyone know if there is a workaround for this? Thank you all regardless. We are loving Photon.

calcmogul, 2021-08-18:
The better solutions are either VMX-Pi doing a 2021 release or PhotonVision making a release targeting WPILib 2020. With that said, a really dirty hack would be making a shim header called `units/time.h` that includes `units/units.h`. It would go in one of the global include paths specified in this spot in build.gradle (this is 3512's, so it may differ for your project):
```groovy
            sources.cpp {
                source {
                    srcDir 'src/main/cpp'
                    include '**/*.cpp', '**/*.cc'
                }
                exportedHeaders {
                    srcDir 'src/main/include'
                }
            }
```
There might be other issues as well though.

MrRSquared, 2021-08-18:
Thank you. That is what I was thinking, but I did not know exactly how to go about creating the shim (which is the only solution we have control over). However, there is another option. I was looking through the docs here and saw that photon is based on Chameleon which only seems to use Network tables for communication which would make it library agnostic on the robot side. I think it would be suitable for our uses, so I am looking into that now. That is probably the best solution until Kuai labs updates to 2021. I am guessing the units/time.h is going to begin a string of errors as I seem to remember the entire units library was overhauled this year.

calcmogul, 2021-08-18:
The header `units/time.h` would look like this:
```cpp
#pragma once

#include "units/units.h"
```
Same goes for any other units headers. All we did was split units.h into a header for each unit, which was a bunch of copy-paste.

Fyi, Chameleon Vision is a dead project. Photon Vision is a rewrite by the core developers over a licensing dispute with the project owner.

bankst-oxos, 2021-08-18:
If you want to use the NetworkTables API, you can look at our documentation for that [here](https://docs.photonvision.org/en/latest/docs/programming/nt-api/nt-api.html)

MrRSquared, 2021-08-18:
Thank you both for your help. I have a lot to work with here, and I think it is enough to get going. I am going to start with the shim file as that will be easier for our members, but we will use the Network Tables API as a fallback. I cannot believe I missed that section of the documentation. I will update here if I make any progress.

calcmogul, 2021-08-18:
What version of photonlib are you using? I know the latest dev release doesn't have windows binaries.

MrRSquared, 2021-08-18:
We are building on Windows and deploying to the bot after disconnecting from the Internet. I will try building while connected to the bot when I get a chance. Our Photon install is on a separate raspberry pi running the latest image. I am not sure the exact version number, but will look when I get a chance. The version of the library is 2021.1.2. Should I downgrade?

calcmogul, 2021-08-18:
That version should work. Could you link your vendordep json? The symbol it's looking for is right here: https://github.com/PhotonVision/photonvision/blob/master/photon-lib/src/main/native/cpp/photonlib/PhotonCamera.cpp, so it should link fine. This usually happens if the linker step just isn't linking to the library at all.

MrRSquared, 2021-08-18:
That did not work either, but I think I found the culprit. 
The vendor dep does not include raspbian binaries (The VMX-Pi OS runs on a Pi). This was not a problem with the Java version, but I wonder if that is the liner error. Here is my current vendor dep. So, maybe we need to use the network tables exclusively (which would be okay). Thank you for your help. 
```
{
    "fileName": "photonlib.json",
    "name": "photonlib",
    "version": "2021.1.2",
    "uuid": "********************************",
    "mavenUrls": [
        "https://maven.photonvision.org/repository/internal"
    ],
    "jsonUrl": "https://maven.photonvision.org/repository/internal/org/photonvision/lib/PhotonLib-json/1.0/PhotonLib-json-1.0.json",
    "jniDependencies": [],
    "cppDependencies": [
        {
            "groupId": "org.photonvision.lib",
            "artifactId": "PhotonLib-cpp",
            "version": "2021.1.2",
            "libName": "Photon",
            "headerClassifier": "headers",
            "sharedLibrary": true,
            "skipInvalidPlatforms": true,
            "binaryPlatforms": [
                "windowsx86-64",
                "linuxathena",
                "linuxx86-64",
                "osxx86-64"
            ]
        }
    ],
    "javaDependencies": [
        {
            "groupId": "org.photonvision.lib",
            "artifactId": "PhotonLib-java",
            "version": "2021.1.2"
        }
    ],
    "required": [
        "https://github.com/PhotonVision/photonvision/blob/master/photon-lib/src/main/native/cpp/photonlib/PhotonCamera.cpp" 
    ]
}
```

calcmogul, 2021-08-18:
Yea, that seems likely. Java can target any platform as long as it doesn't use JNI, but C++ has to be compiled for each platform.

samfreund, 2025-04-08:
Closing as stale

closed 2025-04-08 by samfreund