PhotonVision/photonvision #285: Mismatched Scaling (?) on Stream

gerth2, 2021-09-05:
[image]

I was able to produce a case where the streams of the GPU threshold output and the overlay of target outlines don't match up in the overlay. To make a rough guess, it looks like one is being scaled at a different factor than the other? 

I've not driven to root cause yet.

Settings:

[photonvision-settings.zip](https://github.com/PhotonVision/photonvision/files/7111926/photonvision-settings.zip)

gerth2, 2021-09-05:
Looks tied to using resolutions other than 320*420 on Shape pipeline - Threshold sliders also appear to stop working?

gerth2, 2021-09-05:
`e7035b5de48` in https://github.com/gerth2/photonvision was used to create this, though I don't believe it's specific to that commit.

mcm001, 2021-09-07:
Really weird, can't reproduce this in test mode

mcm001, 2021-09-07:
https://cdn.discordapp.com/attachments/725846784131203222/884780533089964042/PhotonVision_Mozilla_Firefox_2021-09-07_08-41-36.mp4 Nor on my laptop

mcm001, 2021-10-02:
Is this resolved now?

gerth2, 2021-10-02:
I believe yes, on phone and need to look up the PR... but latest on my PC
we had found and corrected root cause

gerth2, 2021-10-02:
Yup - fixed in this commit specifically:
https://github.com/PhotonVision/photonvision/pull/288/commits/ea3e3189eff2248f1abd86a35f8aa36d751efd5e

gerth2, 2021-11-24:
Update: this can be worked around by just showing/hiding/showing the stream. Perhaps post 2022 - dig into these details more and see if we can fix it up

closed 2025-04-02 by Gold856