PhotonVision/photonvision #314: How to set pin to use for LED on Raspberry Pi

viggy96, 2021-11-10:
Its not clear in the documentation how to set which pin to use for LED control on the Raspberry Pi.

gerth2, 2021-11-10:
Here's two examples of the hardware configuration .json:

[gloworm](https://github.com/gloworm-vision/pi-gen/blob/6d81d5f459add3e7f7e2887e064c4b11f39a4c13/stage2/01-sys-tweaks/files/photonvision_config/hardwareConfig.json)
[snakeyes](https://raw.githubusercontent.com/PlayingWithFusion/SnakeEyesDocs/master/PhotonVision/hardwareConfig.json)

So far, the path has been to assume that a hardware vendor or builder would supply these, rather than making it available to the users in a UI.

Were you specifically looking for docs changes to describe the hardware control capabilities and configurations?

viggy96, 2021-11-10:
It would be useful for teams who wish to DIY their own LED solution instead of buying an off the shelf part.

gerth2, 2021-11-22:
For now, going to close this issue as I think it's a documentation thing, not a photon functionality thing? @viggy96 see if you can review the config .json's and make progress, maybe pop open a new issue if you'd like changes to the way LED's are configured?

closed 2021-11-22 by gerth2