PhotonVision/photonvision #315: Raspberry Pi can't open PhotonVision using 802.3-af port on standard OM5P-AC radio. It works when we directly connect ethernet to PC. Should we be using an ethernet switch?

pratyush-p, 2021-11-10:


pratyush-p, 2021-11-10:
Using standard RoboRio

gerth2, 2021-11-22:
@pratyush-p confirming - any code changes needed? or was it just hooking up a network switch on your folks's end?

pratyush-p, 2021-11-22:
Well, my team actually hasn't let me buy it yet....

gerth2, 2021-11-22:
Ha that's a thing. 

If they need motivation [This style](https://www.amazon.com/NETGEAR-Gigabit-Ethernet-Unmanaged-1000Mbps/dp/B00KFD0SMC/ref=sxin_13_ac_d_mf_br?ac_md=1-0-TkVUR0VBUg%3D%3D-ac_d_mf_br_br&cv_ct_cx=ethernet+switch&keywords=ethernet+switch&pd_rd_i=B00KFD0SMC&pd_rd_r=6ceebf17-5108-4b33-965e-8968aed72ab2&pd_rd_w=7fz5D&pd_rd_wg=JHBwa&pf_rd_p=b0c493d8-5fdd-4188-b852-c552a4a3abdb&pf_rd_r=9RP7VHESV61HT08KBDJ3&psc=1&qid=1637547769&sr=1-1-ed8a42d3-65f1-4884-a3a2-0dd6e83b6876) is probably the most common for FRC. Also, [get them for free-ish from FIRST choice](https://firstchoicebyandymark.com/fc-cn-9024)

mdurrani808, 2022-12-30:
Hasn't gotten any updates in a while, happy to reopen if any updates come up.

closed 2022-12-30 by mdurrani808