PhotonVision/photonvision #320: Photon Version Mismatch

MrRSquared, 2021-11-22:
Hello,
I am struggling to get the new version of Photon running right now on new projects. The issue is that I have an older version of Photon on the device (Romi), but when I attempt to connect to it, I get a version of the following error.

`Photon version v2021.1.8 does not match coprocessor version dev!`

I can use the old version on the Romi and modify the vendor lib .Json in my robot program (to v2021.1.7), but the new version is registering as Dev so the current .Json it will not connect to the library. Is this expected behavior?

mcm001, 2021-11-24:
Identified as a CI issue. Again.

mcm001, 2021-11-24:
Hopefully addressed in #328

MrRSquared, 2021-11-24:
Thank you. I am sorry if this was posted elsewhere and I did not see it.

icemannie, 2021-12-05:
Getting the same issue on Gloworm v2021.1.3 running on a Limelight 2. Had another issue in that after flashing the Limelight the hardware on fig.json wouldn’t load.

mdurrani808, 2022-12-01:
Pending further updates, going to close this. Haven't gotten any more reports.

closed 2022-12-01 by mdurrani808