PhotonVision/photonvision #333: Add C++ example code for simposeest

mdurrani808, 2021-11-27:


gaborszita, 2022-12-25:
Will this be implemented in the 2023 photonvision release?

mdurrani808, 2022-12-25:
Hi, we don't currently have C++ code for the simulation pose estimation and don't currently have the bandwidth to write it at the moment. I'll see if anyone has the time. Otherwise, the Java code has all the logic down and should be easy to port to C++.

amquake, 2023-10-10:
> Will this be implemented in the 2023 photonvision release?

Currently, the [apriltagExample](https://github.com/PhotonVision/photonvision/tree/master/photonlib-cpp-examples/apriltagExample) showcases pose estimation and drivetrain simulation, but only supports HIL (hardware-in-the-loop) simulation of PhotonVision, meaning you would need to plug in a physical camera to your computer and run PhotonVision locally.

amquake, 2023-10-10:
Moved to #941

closed 2023-10-10 by amquake