PhotonVision/photonvision #342: Take Snapshot of camera feed from UI

jasondaming, 2021-11-29:
Many times during testing or calibration you want to take a picture of what the camera sees and use that image to tune but it can be hard to keep the camera on that exact scene while experimenting with all of the tuning.  Being able to take a snapshot and then use the new playback feature to tune from that snapshot would be a huge help!

I think this button could either go on the cameras or settings tab but to me it makes more sense on cameras.

mcm001, 2021-12-20:
This should exist with #183 -- if that doesn't work by kickoff we can merge is separately

swalberg, 2022-04-14:
I was thinking it would be handy to be able to view snapshots from the web UI rather than scping. #183 seems to be able to put the bones of that in, too.

While I'm on the topic, the WPILib [DriverStation](https://first.wpi.edu/wpilib/allwpilib/docs/release/java/edu/wpi/first/wpilibj/DriverStation.html) object has information about the current match. I'm not sure if coprocessors have access to that information, but if so, it could be used to give the snapshots a meaningful name if they were taken during an event. e.g. DataLogManager does this to name the files things like `FRC_20220313_203333_DC313_E7.wpilog`.

bankst-oxos, 2022-04-14:
PhotonLib could put that info on to NT from the robot side to make it available to the coprocessor. Definitely open a feature request for that.

swalberg, 2022-04-14:
Thanks, I made #457.

PR #940 (2023-10-17): Take Snapshots from PhotonClient
closes #342 
replace #696
closed 2023-10-17 by srimanachanta