PhotonVision/photonvision #348: Switch Photonlib license

mcm001, 2021-12-02:
Consider switching Photonlib to the Wpilib BSD or MIT licenses?

gerth2, 2021-12-06:
I'm a fan of MIT, but either could work.

From a posturing standpoint, there's not a ton of critical functionality in photonlib, to where I don't see any risk if someone clones it and takes it back into a closed source project . Not really sure why someone would do that anyway.

Bankst, 2021-12-21:
To do this we just need confirmation from all contributors. That list is
@icemannie @mcm001 @gerth2 @prateekma @calcmogul and @Bankst (myself).

I think MIT license is fine for this, but I'm not married to it. BSD would also be fine

icemannie, 2021-12-21:
Either one is fine with me.

calcmogul, 2021-12-21:
You have my permission to use either license.

gerth2, 2022-01-14:
Permission granted

prateekma, 2022-01-15:
You have my permission

Bankst, 2022-01-15:
The license headers in all `photon-lib` source files are still labelled GPL. Will wpiformat correct this with a specific setting?

calcmogul, 2022-01-15:
You specifically excluded all .java files via `licenseUpdateExclude`. If you wanted wpiformat to do something to some .java files, that's the wrong setting.

auscompgeek, 2022-03-19:
This is done as of #395, right?

Bankst, 2022-04-13:
Yes

closed 2022-04-13 by Bankst