PhotonVision/photonvision #354: PhotonCamera Question

icemannie, 2021-12-05:
Have Gloworm 2021-01-11 lite running on a Limelight 2 to a Roborio. Was previously trying to set up Photonvision on a Romi with the same problem.

The photonvision network tables show up in Shuffleboard, and the driver mode and plpeline etries follow the camera web page settings. Problem I'm seeing is that camera.getPipeline() and camera.getDriverMode() don't follow the Network Table. (Only entries tested)

Copying the PhotonCamera code into a new class and changing 

getDriverMode() from return driverMode to return driverModeEntry.getBoolean(false)

works.

mcm001, 2021-12-05:
Thanks! That's odd for sure. It looks like we internally cache the last value the entries were set to -- maybe those are somehow getting out of sync?

icemannie, 2021-12-06:
The NetworkTableEntries for driverMode and pipelineIndex are read into booleans in the PhotonCamera(network table) constructor. So the booleans won't change when their respective entries change. The other methods all use the entries.

mcm001, 2021-12-06:
Yeah, it looks like we assume that the PhotonCamera will be the only thing changing the pipeline/driver mode-iness of the camera. Which Isn't The Case.

icemannie, 2021-12-06:
The setDriverMode needs fixing also. The mode can just be set without reading the current state or use getDriverMode() instead of the boolean driverMode. The boolean can be removed.
Same situation with pipeline index int.

team2194, 2021-12-09:
Found the same issue on setting the led mode. The PhotonCamera file as I am currently using is attached.
[PhotonCamera1.txt](https://github.com/PhotonVision/photonvision/files/7687771/PhotonCamera1.txt)

mcm001, 2021-12-09:
If you'd like feel free to open a PR

mcm001, 2021-12-17:
Should have been addressed by #357

srimanachanta, 2023-12-01:
Are the inconsistencies still present on 2024-beta?

closed 2025-03-20 by spacey-sooty