PhotonVision/photonvision #355: Limelight Install

team2194, 2021-12-09:
The hardware config json file called for in the docs gives the error message "Error while uploading settings file! Could not process provided file."

Exported and unzipped the camera settings, edited the hardwareconfig as attached.

[hardwareConfig.txt](https://github.com/PhotonVision/photonvision/files/7687822/hardwareConfig.txt)

Set "Leds Can Dim" to false since it is a Limelight 2

Now leds will flash and turn off from robot code but won't turn on solid.

mcm001, 2021-12-09:
What PhotonVision version are you using? And the LEDs should be able to dim. That red modal may not be correct, if you look at the logs it appears to work at least on my end

mcm001, 2021-12-09:
I just tested with v2021.1.3, works fine
[image]

icemannie, 2021-12-10:
I used the latest Gloworm 2021.1.3 per the install docs.
This is on a Limelight 2 and the original config file calls for leds can dim to be false which matches the Limelight specification where the right and left led sections can be controlled separately but not dimmed.
[image]

team2194, 2021-12-10:
Started over with gloworm 2021.1.3 flashing he camera with Balena Etcher
Got gloworm.local:5800 in browser and tried the json from the docs with the same result.
Exported the settings zip, unzipped, edited the hardwareconfig, rezipped and imported and the leds work as planned.

Suspect I did something wrong in the rezip before but the same problem still happened with the config from the docs.

closed 2021-12-10 by team2194