PhotonVision/photonvision #369: Convert two assignment statements to the usage of compound operators

elfring, 2021-12-19:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of compound operators](https://docs.oracle.com/javase/specs/jls/se17/html/jls-15.html#jls-15.26 "Description for assignment operators") accordingly.

```diff
diff --git a/photon-core/src/main/java/org/photonvision/vision/pipe/impl/Draw2dTargetsPipe.java b/photon-core/src/main/java/org/photonvision/vision/pipe/impl/Draw2dTargetsPipe.java
index c8e8cb6f..e0764d3e 100644
--- a/photon-core/src/main/java/org/photonvision/vision/pipe/impl/Draw2dTargetsPipe.java
+++ b/photon-core/src/main/java/org/photonvision/vision/pipe/impl/Draw2dTargetsPipe.java
@@ -196,8 +196,8 @@ public class Draw2dTargetsPipe
     }
 
     private void dividePoint(Point p) {
-        p.x = p.x / (double) params.divisor.value;
-        p.y = p.y / (double) params.divisor.value;
+        p.x /= (double) params.divisor.value;
+        p.y /= (double) params.divisor.value;
     }
 
     public static class Draw2dTargetsParams {
diff --git a/photon-core/src/main/java/org/photonvision/vision/pipe/impl/Draw3dTargetsPipe.java b/photon-core/src/main/java/org/photonvision/vision/pipe/impl/Draw3dTargetsPipe.java
index 2313e254..ad37707f 100644
--- a/photon-core/src/main/java/org/photonvision/vision/pipe/impl/Draw3dTargetsPipe.java
+++ b/photon-core/src/main/java/org/photonvision/vision/pipe/impl/Draw3dTargetsPipe.java
@@ -156,8 +156,8 @@ public class Draw3dTargetsPipe
     /** Scale a given point list by the current frame divisor. the point list is mutated! */
     private void dividePointList(List<Point> points) {
         for (var p : points) {
-            p.x = p.x / (double) params.divisor.value;
-            p.y = p.y / (double) params.divisor.value;
+            p.x /= (double) params.divisor.value;
+            p.y /= (double) params.divisor.value;
         }
     }
 
```

Bankst, 2021-12-20:
Thanks for the suggestion. Unless you can prove this was not an automated action, I will close this in 24 hours.

elfring, 2021-12-20:
I dared to present another data processing result from one of my Perl command examples (which I published together with related issues).

-----

The documentation for the programming language “Java” is providing [the following information](https://docs.oracle.com/javase/specs/jls/se17/html/jls-15.html#jls-15.26.2 "Description for compound assignment expressions").

```
A compound assignment expression of the form E1 op= E2 is equivalent to …, except that E1 is evaluated only once.
```

Do in-place data modifications achieve more desirable software run time characteristics at the shown source code places? :thinking:
(Other information sources offer more hints about such implementation details.)

closed 2021-12-20 by Bankst