PhotonVision/photonvision #383: First-time lauch funkyness

gerth2, 2022-01-10:
When running for the first time with the v2022.1.1, on the first launch, the Pi version was detected as "UNKNOWN'. This in turn caused MMAL to not work.

On the second boot, PI_3 was properly detected, but the new pipline was configured with a gain of -1, so the slider didn't show up.

Some time after poking around enough, creating new pipelines, viewing different camera streams, uploading a new .jar ... eventually the gain got set to 50 in the .json, and the slider started showing up.

So, while the problem seems to ultimately be self solving, there's something funky with the first-time launch.

gerth2, 2022-01-10:
Discord convo [starts here](https://discord.com/channels/725836368059826228/725846797339066379/929902975709745182)

mcm001, 2022-01-10:
Does this happen every time at first launch?

closed 2022-01-12 by mcm001

gerth2, 2022-01-12:
Sample size of one says "yes". But that's not useful.

Will pull latest and see if it's reproduced there. Really not the end of the world if it's just a "restart once to get mmal" but will watch for it.

PR #388 (2022-01-12): Ensure cameras with gain will always have it enabled
Closes #383