PhotonVision/photonvision #390: Allow user pipeline settings to be changed over NetworkTables?

mcm001, 2022-01-11:
https://discord.com/channels/725836368059826228/725848198794706994/930439641218289685

> I am thinking about the use case of detecting balls.  Some teams may only want their own ball and it is great for this.  If a team wants either the red or blue ball I see two options:
> 1) expand HSV to include both, will likely increase false positives not sure how much
> 2) Have a red & blue pipeline and constantly switch between them.

Is allowing settings to be changed over NT the best solution? How should that be implemented?

mcm001, 2022-01-12:
Actually, a better solution here would be to lazy-set resolution. That operation takes the longest, and avoiding recreating a Picam every time we switch pipelines would save a lot of time

gerth2, 2022-01-12:
I'd always assumed in a case like this you'd just have "Red Pipeline" and "Blue Pipeline" and switch. But...... This could be nifty.