PhotonVision/photonvision #391: Color Threshold with red is near imposible

1337-haxxor, 2022-01-13:
due to the way using HSV works in photonvision, you would need to set a range of around 165-15 ish for the hue but you are unable to do this so in order to use the color red your hue to all possible values if you hue for you red object like the red cargo balls for 2022.

Bankst, 2022-01-15:
I have thought this over a bit and come up with a possible solution. The idea is to add a toggle to invert the min/max of the range, to allow a hue selection like `330<->15`. This should allow you to capture the red area of the hue circle.

1337-haxxor, 2022-01-15:
I would feel that something like that would fix the issue I'm having.

jonahsnider, 2022-01-29:
This is still a major blocker for vision tracking this year's red cargo. Will this feature be added any time soon?

jsimpso81, 2022-02-07:
I have seen this also issue. One potential solution is to represent the Hue as a circle -- no beginning and ending.  This way the staring hue value can be selected larger than the ending hue value and it can be displayed.  Also the Hue wheel could be shaded to represent the actual Hue.  (This is an example from LabVIEW...)  
[image]

Bankst, 2022-04-13:
Resolved in [v2022.1.5](https://github.com/PhotonVision/photonvision/releases/tag/v2022.1.5)

closed 2022-04-13 by Bankst