PhotonVision/photonvision #400: Packet::DecodeDouble (in Java) can throw an index out of bounds exception

mcm001, 2022-01-16:
https://github.com/PhotonVision/photonvision/blob/master/photon-targeting/src/main/java/org/photonvision/common/dataflow/structures/Packet.java#L151

This can throw if the provided array is too short.

mcm001, 2022-01-16:
Packets should check the array provided matches the expected length, and not do anything if not

mcm001, 2022-01-19:
This is only an issue if people's Photon versions don't match. we should fix anyways

KailenC, 2022-10-15:
when testing the new photonlib with a previously imaged raspberry pi from last season, we ran into this issue. It would be great if this could be fixed.

mdurrani808, 2022-11-02:
The fix for now would be to update so both match if you haven't already. @KailenC

PR #629 (2022-12-09): added checked for packet of incorrect length
This fixes #400 as well as the c++ version of the bug.
closed 2022-12-09 by mcm001