PhotonVision/photonvision #403: Colored shape Reducing Circle Accuracy Slider to 0 causes exceptions every loop

gerth2, 2022-01-16:
STack trace was not put to log either:

```
[2022-01-16 15:57:10] [VisionModule - VisionRunner - USBFrameProvider - opencv_Microsoft_LifeCam_HD-3000] [ERROR] Exception on loop 84804
[2022-01-16 15:57:10] [VisionModule - VisionRunner - USBFrameProvider - opencv_Microsoft_LifeCam_HD-3000] [ERROR] Exception on loop 84805
[2022-01-16 15:57:10] [VisionModule - VisionRunner - USBFrameProvider - opencv_Microsoft_LifeCam_HD-3000] [ERROR] Exception on loop 84806
[2022-01-16 15:57:10] [VisionModule - VisionRunner - USBFrameProvider - opencv_Microsoft_LifeCam_HD-3000] [ERROR] Exception on loop 84807
```

mcm001, 2022-01-19:
Fix is to make the min 1 in the UI

mcm001, 2022-01-21:
We may wanna make a backend fix to resolve this. but this shouldn't happen with the current frontend only fix

mcm001, 2023-01-04:
Closed by #688

closed 2023-01-04 by mcm001