PhotonVision/photonvision #411: problems with running on Ubuntu desktop 16.04 LTS

TrevCan, 2022-01-26:
Hello, I was wondering if you could help me and my team with setting up Photonvision on Ubuntu 16.04. 

I have downloaded the latest stable jar but am getting some trouble running on a Kangaroo Mini PC with Ubuntu 16.04 LTS. Right now I'm upgrading it.

The error I had was with something related to `.wpilib/..../.... GLIBC_2.27 not found...` or something along those lines.

Bankst, 2022-01-26:
Ubuntu 16.04 is not supported by WPILib. You need 18.04 LTS.

TrevCan, 2022-01-26:
Thanks, I think we should add this to the photonvision docs. Didn't know I needed wpilib to run photonvision.

I am in the process of updating to 18.04 LTS. thank you, I will report back.

Bankst, 2022-01-26:
We use WPILib dependencies for camera management, and NetworkTables, along with a few other things. They are packaged in the jarfile, but only target a few select platforms. Ubuntu 18.04 is the 64bit ARM platform that they target.

TrevCan, 2022-01-26:
thanks, Upgrading to Ubuntu 18.04 made it work

closed 2022-01-26 by TrevCan