PhotonVision/photonvision #412: Automated script not working.

TrevCan, 2022-01-27:
Hello, first of all thanks for making this.

I recently installed Ubuntu Bionic 18.04.06 LTS.

Then I proceeded to run the automated script, recommended to use with Ubuntu bionic: https://docs.photonvision.org/en/latest/docs/getting-started/installation/coprocessor-image.html#other-debian-based-co-processor-installation

Then i tried to start the service but it failed. Log of systemd service ( after trying to start it)

```bash
root@kangaroo:~# systemctl status photonvision
WARNING: terminal is not fully functional
● photonvision.service - Service that runs PhotonVision
   Loaded: loaded (/etc/systemd/system/photonvision.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2022-01-26 19:04:44 CST; 114ms ago
  Process: 7019 ExecStart=/usr/bin/java -jar /opt/photonvision/photonvision.jar (code=exited, status=1/FAILURE)
 Main PID: 7019 (code=exited, status=1/FAILURE)
```

Running the command manually from `photonvision.service`, yields the following:
```bash
root@kangaroo:/opt/photonvision# /usr/bin/java -jar /opt/photonvision/photonvision.jar
Error: Invalid or corrupt jarfile /opt/photonvision/photonvision.jar
```

then I tried manually downloading the latest stable release:
```bash
kangaroo:~/photonvision$ wget "https://github.com/PhotonVision/photonvision/releases/download/v2022.1.4/photonvision-v2022.1.4.jar"
```

then tried running it manually with
```bash
baby@baby-kangaroo:~/photonvision$ java -jar photonvision-v2022.1.4.jar
```

**and it worked!** but not using the systemd service. Any thoughts ? thanks.

Bankst, 2022-01-27:
Best I can tell is that the download just happened to be corrupt, or the script grabbed an "empty" jar. I've seen this happen before, but since very few people use that script, it's never come up as a concern.

If you can reproduce this and get the SHA1 hash (`sha1sum myfilename`) of the "corrupt" jar and of the 2022.1.4 jar and post them here, that'd be helpful.

TrevCan, 2022-01-27:
Of course. 

```
baby@baby-kangaroo:~/photonvision$ sha1sum /opt/photonvision/photonvision.jar
da39a3ee5e6b4b0d3255bfef95601890afd80709  /opt/photonvision/photonvision.jar
```

Bankst, 2022-01-27:
Is that the "corrupt" jar?
for 2022.1.4 I get `14e28c6c59abecc1a07ae442535b8395b298789e`

TrevCan, 2022-01-27:
yes, it is the one 'automatically' installed. for the one I downloaded, also 2022.1.4 i get `14e28c6c59abecc1a07ae442535b8395b298789e `

So yeah, probably corrupt. I'll try re-running the script later. For now, vision works!

I guess it wouldn't be hard settting up a systemd service on startup for the manually installed one.

TrevCan, 2022-01-27:
Oh, I just ran it again and guess what ? I don't have `curl` installed. That's definitely the culprit.

TrevCan, 2022-01-27:
I guess maybe using wget is more universal ?

Bankst, 2022-01-27:
Ah. Our script is somewhat naive and wouldn't error on `curl` being missing. `wget` should work fine in its place.

TrevCan, 2022-01-27:
Thanks. Although one important thing: it downloads the **latest release. Not the latest and stable.**

Bankst, 2022-01-27:
`latest` is always considered stable, as it doesn't count pre-release releases.

TrevCan, 2022-01-27:
Oh, interesting. 

Interestingly enough, it doesn't match with the sha1s we shared. 

The phtoonvision service now works! (After deleting some lines, since it doesn't delete/clear the service file before installing.

the sha1 i'm getting is: `ac3c33c74defa8cf3b7607f63434cb256dac8f1c`.

closed 2022-04-13 by Bankst