PhotonVision/photonvision #414: Enhancement --  Add PhotonVisionLib for LabVIEW

jsimpso81, 2022-02-04:
Enhancement  -- After reading a number of posts on chiefdelphi about interest in PhotonVisionLib for LabVIEW, I ported the Camera, Utility and some ther functions to LabVIEW.  Would you like take over the library and host it here?  (Please...)   I'd be happy to continue working on it if that is desirable.

The library is here:
https://github.com/jsimpso81/PhotonVisionLabVIEW

Thanks

jsimpso81, 2022-02-12:
Updated the external github repository.  The library now implements the photonlib test programs.  Some additional convenience functions were also added.

Bankst, 2022-04-13:
We need to update documentation to reference this. @mdurrani808

spacey-sooty, 2025-03-02:
Do we wanna close this with the move off labview for 2027?

samfreund, 2025-04-15:
> Do we wanna close this with the move off labview for 2027?

I think wait until 2027 if someone feels like doing it for 2026? It would be kinda a waste, but idk.

mcm001, 2025-04-17:
We could consider monorepoing labview for 2026, but I think you'd be the only person who could maintain it (none of our other current contributors have bandwidth to), and LabView is going to be sunset in favor of Python+Blocky in 2027. I'm going to close as unplanned for now, but please don't take that as a lack of interest from our side! Really appreciate everything youve done for the labview ecosystem in FRC.

closed 2025-04-17 by mcm001