PhotonVision/photonvision #417: Static IP address interesting behaviour

jsimpso81, 2022-02-07:
My architecture:

    Raspberry PI 3b - PhotonVision image 1.4
    Latested 2022 updated FRC game tools
    1 USB camera
    3 pipelines (almost identical - orange ball, red ball, blue ball detect) Robot can switch between pipelines.

I used the web interface to set a static IP address.   Turned off DHCP and entered 10.TE.AM.10  (replace TEAM with team number).    Saved this configuration and restarted the Raspberry PI.    The static IP address showed up and worked.  A static IP address also showed up and worked.    Cycled power on raspberry PI again.  This time the static IP address didn't show up.  Repeatedly rebooting (power cycling) the Raspberry PI seems to toggle this behavior.

A workaround is that the camera stream URL have the IP address in the second streaming string.  This can be seen on the Dashboard.
It might be nice to have a separate network table value written with just the current IP address to make it easier to find.

It would be nice to have a consistent static IP address.

gerth2, 2022-02-07:
@dkt01 had noticed something similar, but I've not yet been able to reproduce on my Pi. 

Granted, it is on my home network, which is on Ubiquity hardware, not the same as a robot.

tom131313, 2022-11-14:
photonvision-v2023.1.1-beta-6-raspi-image.xz and the .jar

When I select Static the SAVE button is grayed out and static is not saved. Only DHCP has an active SAVE button. Happens every time. Setup is driver station PC, radio, roboRIO, RPi 4b.