PhotonVision/photonvision #427: CameraServer is lost on code deploy to roborio

bruingineer, 2022-02-20:
Everytime we deploy code to the roboRIO, the CameraServer does not show up in network tables. After a power cycle of the robot, the camera server shows up again.

Has anyone seen this behavior before?

in robotInit():
```
var cameraName = "mmal_service_16.1";
camera = new PhotonCamera(cameraName);
```

bruingineer, 2022-02-20:
I solved this by adding 
```
var cameraName = "mmal_service_16.1";
if (camera == null)
        camera = new PhotonCamera(cameraName);
```
to robotPeriodic() or similar

bruingineer, 2022-02-22:
We are still getting 

```
PhotonVision coprocessor at path /photonvision/mmal_service_16.1 not found on NetworkTable!
```
in our robot log when we deploy code to the robot.

It resolves itself on a robot power cycle.

legoguy1000, 2022-03-07:
We have also experienced this issue as well.

trevnels, 2022-03-14:
We're seeing this as well. Restarting photon or unplugging/replugging our coprocessor seems to be a working temporary fix.

srimanachanta, 2023-12-01:
Is this issue still happening?

> ```
> var cameraName = "mmal_service_16.1";
> if (camera == null)
>         camera = new PhotonCamera(cameraName);
> ```
> 
> to robotPeriodic() or similar

I would need to see more of your code implementation to trace an issue here. It could be in your robot code or the backend not publishing data.

Gold856, 2025-03-20:
Have not seen issues in 2025 release. If this is still an issue, feel free to reopen.

closed 2025-03-20 by Gold856