PhotonVision/photonvision #449: Don't scream at me when PhotonLib and PhotonVision patch versions are not equal

MarkGhebrial, 2022-03-26:
PR #302 causes PhotonLib to throw errors when its version is out of sync with the PV version running on the coprocessor. However, it's safe to assume that no breaking change would occur between *patch* revision numbers.

For example, PhotonLib v2022.1.*6* prints a huge stack trace every second or so when I'm using PhotonVision v2022.1.*5*, which is not only annoying, but causes my robot code to exceed the 20ms loop time.

I'd appreciate any of the following changes:
1. Make PhotonLib ignore patch versions as long as the year and minor versions are OK
2. Allow the user of PhotonLib to turn off version mismatch warnings
3. Don't print the entire stack trace

auscompgeek, 2022-03-27:
The version check can be disabled with PhotonCamera::SetVersionCheckEnabled.

Bankst, 2022-03-27:
Another thing we can do is only print it once instead of every loop.

mcm001, 2023-01-01:
Closing as we have a way right now to disable and limit the number of warnings.

closed 2023-01-01 by mcm001