PhotonVision/photonvision #454: Pipeline not being loaded correctly under v2022.2.0

viggy96, 2022-04-08:
Our team only has a single vision pipeline, called "default". We've tuned it reasonably well, so that it identifies Rapid React targets. However, sometimes when we disable driver mode in the code, it appears to not load our pipeline, since the vision output is highlighting a bunch of stuff it shouldn't be. Going to the web UI and toggling driver mode on and off fixes the issue, but is very annoying and concerning for use in competition.

Bankst, 2022-04-08:
This is a purely visual bug and does not impact targeting data

viggy96, 2022-04-11:
I'm not sure this is a purely visual bug. During our state competition, we had to toggle driver mode on/off before every match to make sure our vision was working correctly.

Bankst, 2022-04-13:
How did you verify that the tracking data was affected by this? My team experienced the very same bug and saw no change in tracking data whether or not we toggled the stream to resolve the visual glitch.

viggy96, 2022-04-21:
Just tried not toggling during a match at worlds. Not only did targeting not work, the other webcam stream wasn't working as well.

viggy96, 2022-04-22:
Not during a match, I can go to the PhotonVision webpage, and I don't get targeting data showing up until I toggle on/off.

viggy96, 2022-04-26:
the `getBestTarget()` method returns null as well.

mdurrani808, 2022-05-01:
@viggy96 Do you have steps to reproduce this issue? I'll have some time to look into it.

viggy96, 2022-05-01:
@mdurrani808 Not really any steps to speak of. I bootup the robot (SnakeEyes Pi attached), while pointed at a target. I enable the robot (which disables driver mode in my code), and I get no valid targets. I go to the PhotonVision webpage, and I still get no targets until I toggle driver mode on/off.

mdurrani808, 2022-05-01:
Got it. A link to your code would be helpful as well if it is public, I'll see if I can reproduce it.

viggy96, 2022-09-19:
Sure. https://github.com/FRC4561TerrorBytes/TB2022

viggy96, 2022-09-19:
Has anyone been able to reproduce this?

mdurrani808, 2022-09-19:
Sorry, I have not been able to reproduce it and have lost robot access.

mdurrani808, 2022-09-19:
One thing I would try: it looks like you have a lot of hardware based abstraction going on at the moment, I would try the following: 
1. Reflash your pi, just for good measure
2. Retune your pipeline to your liking, and then save it.
3. Set a static ip. This seems to fix a variety of issues.
4. Make a minimum viable program/subsystem file for the vision system (ex. no hardware abstraction, directly accessing the photon camera for values, etc. just barebones that would get you values) to see if there is a problem somewhere in the abstraction. I doubt this personally, but I've had it happen on our team. Follow the steps of 
5. Using Glass or whatever dashboard, put a target in front of the camera, wave it around, and make sure values update correctly. Then, toggle driver mode on and off, make sure that all works. Powercycle, try again, just to confirm.
6. https://github.com/FRC4561TerrorBytes/TB2022/blob/master/src/main/java/frc/robot/subsystems/VisionSubsystem.java#L160-L161 You shouldn't get the bestTarget before you even know if you have valid targets.

Sorry for this being a lot, but with it being hard to reproduce, not much I would know how to help. Let me know if you're interested in trying these steps are report back if so!

viggy96, 2022-09-19:
3. The Pi already has a static IP.
6. Worst case scenario bestTarget returns null, in which case that's taken care of by the following if statement.

I'll report to my team, and try to see if this resolves anything.

Lann4daze, 2022-09-28:
> Has anyone been able to reproduce this?

My team, FRC 1225 is currently having this same issue, where we duplicated our working pipeline as a backup, and it now refuses to switch to said backup. I downloaded the logs and the logs have a WARN message, “User attempted to set index to non-existent pipeline!” We are using a limelight 2+ running photonvision v2022.2.0

Lann4daze, 2022-09-28:
I just checked the cameras/mmal_service_16.1/pipelines, and it has the JSON file for the backup intact, with an index that is unique

thegorillas1225, 2022-09-28:
attached is the exported zip from the photonvision install

[photonvision-settings (2).zip](https://github.com/PhotonVision/photonvision/files/9669530/photonvision-settings.2.zip)

srimanachanta, 2023-12-01:
Is this issue still happening on 2024-beta-3.2?

Gold856, 2025-04-12:
I tested this under 2025.1.1 and switching driver mode on and off seems to work fine. Closing as stale.

closed 2025-04-12 by Gold856