PhotonVision/photonvision #455: Weird issue when two Lifecams are connected

AJahueyM, 2022-04-09:
https://discord.com/channels/725836368059826228/725848198794706994/962161501324574730
Here are the logs showing the issue.
[image]
[image]

Here are the logs we get when we restart the service
[image]

BerkeSinanYetkin, 2022-10-08:
Has anyone found a solution for this? we are having the same issue.

gerth2, 2022-10-30:
Critically - this one has not yet been worked because no one on the current development team has two lifecams at their disposal.

If anyone would like to send us two lifecams (or needs help getting started on the debugging and you have two lifecams), let us know.

realSaddy, 2022-10-31:
I set up a Raspberry PI 3B with two HD 3000 Lifecams on the latest beta release. I was not able to replicate this error message; everything worked without issue.

AJahueyM, 2023-12-01:
I'm not sure, we are no longer using Lifecams for the robot. This was posted a while ago.

samfreund, 2025-04-14:
Closing as stale, and cannot reproduce.

mcm001, 2025-04-14:
we no longer use mmal anyways

closed 2025-04-14 by Gold856