PhotonVision/photonvision #465: Create a more robust way of handling toggling LEDs

realSaddy, 2022-05-08:
The current code only runs LEDs off of a camera if it is a picam. If one uses a USB camera with the current code, the LEDs will not be controllable.

A better solution should be created for this.

samfreund, 2025-08-13:
Closing as LEDs are not generally in use

closed 2025-08-13 by samfreund

mcm001, 2025-08-13:
GPIO control is still a thing we might want to better support. Are there any other tracking issues for it? If not, let's reopen or officially drop gpio support entirely :P

Gold856, 2025-08-14:
Let's drop it. Since people are mostly using Orange Pi 5's and Arducams now, LEDs aren't really being used anymore. I could see a case for using the GPIO to trigger frame capture and therefore get even more accurate frame timestamps, but I doubt it would be widely adopted (lot of work for not that much gain).