PhotonVision/photonvision #481: PhotonVision on NVidia Jetson

agasser, 2022-10-03:
The [installation documentation](https://docs.photonvision.org/en/latest/docs/getting-started/installation/coprocessor-image.html) suggests PhotonVision will run on an NVidia Jetson, but it does not work.

First, the installation script requires `curl`, but it is not pre-installed on the Jetson. An easy workaround is to manually install it first:
```
sudo apt update
sudo apt install curl
```

Once PhotonVision is running successfully, it is not able to use the attached USB camera. The following output prints repeatedly:
```
CS: WARNING: vi-output,_ov5693_2-0036: invalid JPEG image received from camera (UsbCameraImpl.cpp:528)
```

I attempted this with a Jetson TX2 using a USB ELP camera.

mcm001, 2022-10-05:
If you could send log files that would be helpful, but based on this it sounds like CSCore doesn't like the USB camera you used

agasser, 2022-10-06:
I think you're onto something. It looks like it's actually the camera onboard the Jetson TX2 that cscore doesn't like. When I try using cscore with python and that camera I get the same message.

However, it doesn't matter which camera I select, I am unable to get a stream to appear.

Here is the log output:
```
[2022-10-05 22:04:51] [General - Main] [ERROR] Failed to parse command-line options!
[2022-10-05 22:04:51] [General - Main] [INFO] Logging initialized in debug mode.
[2022-10-05 22:04:51] [General - Main] [INFO] Starting PhotonVision version v2022.2.0 on Unknown Platform. OS: Linux, Architecture: aarch64
[2022-10-05 22:04:51] [General - Main] [INFO] Native libraries loaded.
[2022-10-05 22:04:51] [General - ConfigManager] [INFO] Loading settings...
[2022-10-05 22:04:51] [General - ConfigManager] [INFO] Hardware config does not exist! Loading defaults
[2022-10-05 22:04:51] [General - ConfigManager] [INFO] Hardware settings does not exist! Loading defaults
[2022-10-05 22:04:51] [General - ConfigManager] [INFO] Network config file does not exist! Loading defaults
[2022-10-05 22:04:51] [General - ConfigManager] [DEBUG] Cameras config folder did not exist. Created!
[2022-10-05 22:04:51] [General - NetworkTablesManager] [INFO] Starting NT Client
[2022-10-05 22:04:51] [General - NetworkTablesManager] [ERROR] [NetworkTablesManager] Could not connect to the robot! Will retry in the background...
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [DEBUG] Matching 3 new cameras!
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [INFO] Adding local video device - "H264 USB Camera" at "/dev/video0"
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [INFO] Adding local video device - "H264 USB Camera" at "/dev/video1"
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [INFO] Adding local video device - "vi-output, ov5693 2-0036" at "/dev/video2"
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [DEBUG] After matching loaded configs 3 cameras were unmatched.
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [INFO] Creating a new camera config for camera H264_USB_Camera
[2022-10-05 22:04:52] [Camera - CameraConfiguration] [DEBUG] Creating USB camera configuration for UsbCameraH264 USB Camera (AKA H264_USB_Camera) at /dev/video0
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [INFO] Creating a new camera config for camera H264_USB_Camera (1)
[2022-10-05 22:04:52] [Camera - CameraConfiguration] [DEBUG] Creating USB camera configuration for UsbCameraH264 USB Camera (AKA H264_USB_Camera (1)) at /dev/video1
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [INFO] Creating a new camera config for camera vi-output,_ov5693_2-0036
[2022-10-05 22:04:52] [Camera - CameraConfiguration] [DEBUG] Creating USB camera configuration for UsbCameravi-output, ov5693 2-0036 (AKA vi-output,_ov5693_2-0036) at /dev/video2
[2022-10-05 22:04:52] [Camera - VisionSourceManager] [DEBUG] Matched or created 3 camera configs!
[2022-10-05 22:04:52] [WebServer - DataChangeService] [DEBUG] Added subscriber - Sources: DCS_WEBSOCKET, DCS_HTTP, DCS_NETWORKTABLES, DCS_VISIONMODULE, DCS_OTHER, Destinations: DCD_UI
[2022-10-05 22:04:52] [WebServer - DataChangeService] [DEBUG] Added subscriber - Sources: DCS_WEBSOCKET, Destinations: DCD_GENSETTINGS
[2022-10-05 22:04:53] [Camera - USBCameraSource - H264_USB_Camera (1)] [ERROR] Got a null video mode! Doing nothing...
[2022-10-05 22:04:53] [Camera - USBCameraSource - vi-output,_ov5693_2-0036] [ERROR] Got a null video mode! Doing nothing...
[2022-10-05 22:04:53] [Camera - VisionSourceManager] [DEBUG] Matched config for camera "USBFrameProvider - opencv_H264_USB_Camera" and loaded 0 pipelines
[2022-10-05 22:04:53] [Camera - VisionSourceManager] [DEBUG] Matched config for camera "USBFrameProvider - opencv_H264_USB_Camera (1)" and loaded 0 pipelines
[2022-10-05 22:04:53] [Camera - VisionSourceManager] [DEBUG] Matched config for camera "USBFrameProvider - opencv_vi-output,_ov5693_2-0036" and loaded 0 pipelines
[2022-10-05 22:04:53] [Camera - VisionSourceManager] [INFO] Adding 3 configs to VMM.
[2022-10-05 22:04:53] [General - NetworkTablesManager] [ERROR] NT Connection has failed! Will retry in background.
[2022-10-05 22:04:53] [WebServer - DataChangeService] [DEBUG] Added subscriber - Sources: DCS_WEBSOCKET, DCS_HTTP, DCS_NETWORKTABLES, DCS_VISIONMODULE, DCS_OTHER, Destinations: DCD_ACTIVEMODULE, DCD_ACTIVEPIPELINESETTINGS, DCD_GENSETTINGS, DCD_UI, DCD_OTHER
[2022-10-05 22:04:53] [VisionModule - VisionModule - H264_USB_Camera] [INFO] Setting pipeline to 0
[2022-10-05 22:04:53] [Camera - USBCameraSource - H264_USB_Camera] [DEBUG] Setting camera exposure to 50
[2022-10-05 22:04:53] [WebServer - DataChangeService] [DEBUG] Added subscriber - Sources: DCS_WEBSOCKET, DCS_HTTP, DCS_NETWORKTABLES, DCS_VISIONMODULE, DCS_OTHER, Destinations: DCD_ACTIVEMODULE, DCD_ACTIVEPIPELINESETTINGS, DCD_GENSETTINGS, DCD_UI, DCD_OTHER
[2022-10-05 22:04:53] [VisionModule - VisionModule - H264_USB_Camera (1)] [INFO] Setting pipeline to 0
[2022-10-05 22:04:53] [Camera - USBCameraSource - H264_USB_Camera (1)] [ERROR] Got a null video mode! Doing nothing...
[2022-10-05 22:04:53] [Camera - USBCameraSource - H264_USB_Camera (1)] [DEBUG] Setting camera exposure to 50
[2022-10-05 22:04:53] [WebServer - DataChangeService] [DEBUG] Added subscriber - Sources: DCS_WEBSOCKET, DCS_HTTP, DCS_NETWORKTABLES, DCS_VISIONMODULE, DCS_OTHER, Destinations: DCD_ACTIVEMODULE, DCD_ACTIVEPIPELINESETTINGS, DCD_GENSETTINGS, DCD_UI, DCD [... cut ...]
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:829)

[2022-10-05 22:04:53] [WebServer - SocketHandler] [INFO] New websocket connection from /192.168.1.215:52707
[2022-10-05 22:04:54] [General - HardwareManager] [INFO] Shutting down LEDs...
```

mdurrani808, 2022-10-13:
@agasser https://www.chiefdelphi.com/t/issues-with-viewing-a-usb-camera/343005/9

mdurrani808, 2022-10-13:
See note about logitech cameras here: https://docs.photonvision.org/en/latest/docs/hardware/supportedhardware.html#supported-cameras

agasser, 2022-10-13:
> @agasser https://www.chiefdelphi.com/t/issues-with-viewing-a-usb-camera/343005/1
> Please read above w/rt the ELP camera error.

This looks like guidance for correcting the error in the PhotonVision code, not something I can do as a user, unless I'm missing something. The ELP camera does work on the Jetson with cscore.

This python cscore code works with the ELP camera. It is setting the mode and resolution/fps like Peter suggested.
```
frontCam = cscore.UsbCamera("DriverCam0", "/dev/v4l/by-path/platform-70090000.xusb-usb-0:3.2:1.0-video-index0")
frontCam.setPixelFormat(cscore.VideoMode.PixelFormat.kMJPEG)
frontCam.setResolution(320, 180)
frontCam.setFPS(20)
```

As a user PhotonVision user, is there something I can do?

agasser, 2022-10-13:
> See note about logitech cameras here: https://docs.photonvision.org/en/latest/docs/hardware/supportedhardware.html#supported-cameras

Yes, that is a problem. In conflict with the link you shared, my experience is that the Logitech C525 _does_ work, the ELP camera _does not_.

mdurrani808, 2022-10-13:
> > See note about logitech cameras here: https://docs.photonvision.org/en/latest/docs/hardware/supportedhardware.html#supported-cameras
> 
> Yes, that is a problem. In conflict with the link you shared, my experience is that the Logitech C525 _does_ work, the ELP camera _does not_.

Ahh, I was misreading what was above. Sorry about that.

mdurrani808, 2022-10-25:
Still doing some digging into this. Have a link to the ELP camera?

mcm001, 2022-10-25:
If the MJpeg format is a known fix, the best solution may be adding a camera quirk in photon code

agasser, 2022-10-25:
> Still doing some digging into this. Have a link to the ELP camera?

Here it is https://a.co/bgCAFAK

mdurrani808, 2022-10-27:
Are you able to print out all the current video modes using your Python code? Wondering if all of them just register as kUnknown and the list ends up null.
https://robotpy.readthedocs.io/projects/cscore/en/stable/_modules/cscore/cameraserver.html
[image]

agasser, 2022-10-28:
I added this code to the `USBCameraSettables` constructor and [this](https://github.com/PhotonVision/photonvision/files/9890943/camera.modes.txt) is what came out.
```
            System.out.println(
                    "--------CAMERA: "
                            + configuration.baseName
                            + " "
                            + configuration.path
                            + " modes:"
                            + videoModes.size());
            for (VideoMode vm : videoModes.values()) {
                System.out.println("h: " + vm.height);
                System.out.println("w: " + vm.width);
                System.out.println("fps:" + vm.fps);
                System.out.println("fmt: " + vm.pixelFormat.name());
                System.out.println("--------");
            }
```

It sees the USB camera as two cameras, `/dev/video0` and `/dev/video1`. `/dev/video2` is the onboard csi camera. `/dev/video0` does have 70 modes.

In the PhotonVision UI I see two cameras "H264_USB_Camera" and "H264_USB_Camera (1)". Both cameras seem to list all of the modes. Neither camera will create a stream in the UI.

mcm001, 2022-10-28:
Odd that it shows up as two cameras (/dev/video0 and /dev/video1)... maybe that's the problem? We should be ignoring the fake camera at video1, but we don't? Just a theory

agasser, 2022-10-28:
This might be helpful, here is some output from v4l2-ctl. Notice how `/dev/video1` only exposes H.264 compressed formats.
```
$ v4l2-ctl --list-devices
vi-output, ov5693 2-0036 (platform:15700000.vi:2):
        /dev/video2

H264 USB Camera (usb-3530000.xhci-2):
        /dev/video0
        /dev/video1
```

```
$v4l2-ctl --device=/dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : Motion-JPEG
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)

        Index       : 1
        Type        : Video Capture
[... 71 lines cut ...]
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
```

agasser, 2022-10-28:
Also, the original message I noted, `CS: WARNING: vi-output,_ov5693_2-0036: invalid JPEG image received from camera (UsbCameraImpl.cpp:528)`, is coming from the built in csi camera. If I add a `CompletelyBroken` quirk for this camera those errors stop. That's a different issue.

mdurrani808, 2022-10-28:
So it reads the devVideo 1, sees all are unknown (h264), gets a null video mode, and then leads to those errors. Now to figure out why it won't even display a stream at all for either camera..I agree that the fake camera, or any camera with null video modes should be ignored or elegantly dealt with..

mcm001, 2022-10-28:
Ideally, we don't let a VisionModule even get created for a camera that doesn't have any video modes at all I think?

mdurrani808, 2022-10-29:
Lol, found this from earlier this year: https://www.chiefdelphi.com/t/photonvision-cant-find-usb-camera/401159

Might look into how WPILib pi handles usb cameras?

agasser, 2022-10-29:
> Lol, found this from earlier this year: https://www.chiefdelphi.com/t/photonvision-cant-find-usb-camera/401159

That sure looks like the same problem.

I tried hacking the blacklist to exclude the camera that comes up with no compatible modes and the camera worked. It didn't seem like it was working at first but I toggled driver mode and auto-exposure a couple times and it started working.

I'll play around more and see if I can come up with something reliable that could be the base for a pull request.

agasser, 2022-10-29:
#550 nearly fixes the issue. The PR causes the built-in CSI camera and the second H.264 compressed camera to be ignored.
The only remaining issues is if I delete the config, the camera does not stream until I toggle auto-exposure.

mdurrani808, 2022-10-30:
Which config are you talking about? Thank you for the PR by the way, much appreciated!

agasser, 2022-10-30:
> Which config are you talking about? Thank you for the PR by the way, much appreciated!

The saved config files on the server. I tried that just to get a fresh start.

PR #550 (2022-10-31): Ignore cameras with no supported VideoModes #481
Fixes #481
closed 2022-10-31 by mcm001