PhotonVision/photonvision #505: ELP Cameras change binning based on image type

gerth2, 2022-10-19:
This messes up our strategy of applying calibrations to all resolutions

https://www.amazon.com/180degree-Fisheye-Camera-usb-Android-Windows/dp/B00LQ854AG/ was specifically what I exposed this on.

800x600 MJPEG:
[image]

800x600 YYUV:
[image]

The YYUV binning shifts off to one size, while the MJPEG is centered

gerth2, 2022-11-03:
Not 100% sure what to do about this one. Perhaps a new quirk, then exclude the non-centered resolutions?

mcm001, 2023-01-04:
Quirk makes sense to me. That's really curst behavior.

samfreund, 2025-04-14:
Closing as duplicate of #1168