PhotonVision/photonvision #516: simposeest should use 3D until addVisionMeasurement

agasser, 2022-10-20:
The [simposeest](https://github.com/PhotonVision/photonvision/blob/bfc5e45cd0aa3f6ab6002ef40ff754716c07f29d/photonlib-java-examples/src/main/java/org/photonlib/examples/simposeest/robot/DrivetrainPoseEstimator.java#L91) example converts `Transform3d` returned from `getCameraToTarget` to `Trasnform2d` and then works with 2D. The recommended approach would be to use 3D and only convert to 2D at the end before calling `addVisionMeasurement`.

agasser, 2022-10-20:
I can put together a PR for this, once I get a chance to test it out on a real robot.

gerth2, 2022-10-20:
Thanks! Yup this is correct. While the final architecture is still TBD, the work in flight is here - https://github.com/PhotonVision/photonvision/pull/512