PhotonVision/photonvision #520: Add UI Notification when a broken camera is detected

gerth2, 2022-10-21:
Certain cameras have issues when cscore interacts with them. These are currently marked with the `COMPLETELY_BROKEN` quirk. 

Add some UI notification to users when a `COMPLETELY_BROKEN` camera is detected so they know it's found, but being ignored.

gerth2, 2023-01-26:
Bump. Lots of users seem to still hit this.

Additionally, add support to detect when the camera isnt supplying frames, and put that in the UI as well.

srimanachanta, 2023-12-01:
If we add metadata to all cameras with these props we only emit on full settings structs we could add this, specifically camera quirks and other relevant data to disabling a camera. Would require making a vision module though, might be easier just to send dead cameras seperate.

martincarapia, 2024-04-07:
So here, adding a toast notification saying something like, "Camera <identifier> detected, but ignored." Identifier can be the Camera name or some sort of identifying meta data. Would that be sufficient?

mcm001, 2025-03-20:
We could consider indicating this in quirk in the new camera matching UI?