PhotonVision/photonvision #536: Missing alt for Camera Calibration video img source

tom131313, 2022-10-25:
Often the video stream does not appear in the [Camera Calibration frame](http://10.42.37.11:5800/#/cameras). Manually using browser Load image often makes the image appear. Unfortunatley there is no cue as to where the image should be thus making it hard to load manually. Please add the **alt="Stream1"** like the Dashboard image.

Cameras CAMERA CALIBRATION:
[image]

Dashboard PROCESSED:
[image]

gerth2, 2022-10-25:
https://github.com/PhotonVision/photonvision/pull/529 and https://github.com/PhotonVision/photonvision/pull/422 may also be of interest to help address root cause

gerth2, 2022-11-03:
Tentatively closing this as I expect the websocket streaming to cover this up. Feel free to re-open if needed.

closed 2022-11-03 by gerth2