PhotonVision/photonvision #537: Decimate 2.5 wont detect tags

Technologyman00, 2022-10-26:
if I set Decimate to 2.5 it wont detect tags. 0, 0.5, 1, 1.5, 2, and 3 all work.

realSaddy, 2022-11-03:
The bug is in https://github.com/PhotonVision/apriltag/blob/master/apriltag.c#L815 line 815. Whenever decimate is set to 2.5, g1 and g2 are zero, thus weight is zero, and thus the loop fails with zero division. If it makes since to have a 2.5 decimate (which I'm not sure of), then these lines would need to be fixed https://github.com/PhotonVision/apriltag/blob/master/apriltag.c#L805-L806.

gerth2, 2022-11-03:
If 2.0 and 3.0 are valid, yea, 2.5 ought to make sense. I gotta stare at the code a bit more.... what's special about 2.5 in that case that leads g1 and g2 to be guarnteed to be zero ?

realSaddy, 2022-11-03:
It's not just 2.5, 2.3 also suffers from the same problem from what I've tested.

gerth2, 2022-11-03:
Interesting. Restricting to integer decimation is probably reasonable. Fractional is technically possible but I'd assume more resource intensive? Assuming my mental model of the world is correct.

realSaddy, 2022-11-04:
Agreed, but .5 & 1.5 are valid and working decimates. Could change it to use HTML range datalist tickmarks with JavaScript to step down/up to the nearest value? Or, simpler, change the input to a drop-down.
 
My bet is the pixel centering doesn't work nicely.

gerth2, 2022-11-08:
Hmm. I have to wonder, based on code inspection, if the 0.5 and 1.5 are actually doing interpolation, or just happen not to crash.

closed 2022-11-10 by gerth2