PhotonVision/photonvision #538: Default name for new pipeline is not automatically incremented

laviRZ, 2022-10-26:
The expected behavior from the user standpoint is when creating a new pipeline the placeholder value will be a valid one, such as "New Pipeline 2" and 3 after that and so on.

mdurrani808, 2022-10-26:
Should be easy if you'd be interested in contributing it:
https://github.com/PhotonVision/photonvision/blob/87e7c3ca746402f701d2d20a2497061a6f97c360/photon-core/src/main/java/org/photonvision/vision/pipeline/CVPipelineSettings.java#L42
That line seems to be the relevant one, just a "New Pipeline " + pipelineIndex should do it

samfreund, 2025-04-12:
Closing as no longer relevant.

see [here](https://github.com/PhotonVision/photonvision/pull/1012#issuecomment-2798417991) for more details.

closed 2025-04-12 by samfreund