PhotonVision/photonvision #552: Robustly remember previous pipeline, resolution, etc. between system shutdowns

mdurrani808, 2022-10-30:
Based on #271 and reports on CD, pipeline and resolution could change between start up. More digging to be done

gerth2, 2022-10-31:
Tentative thoughts...

1) Crank up the priority/frequency of the save-to-disk
2) Add some form of a watchdog mechanism to check whether things are saved to disk, or at least flag when a save hasn't happened in a while?
3) Add a UI thingy which tells the user how long its been since last save. This won't actually fix anything, but just make it easier for end users to make reproducible test cases where they change settings, but _don't_ see a save occur.

srimanachanta, 2023-12-01:
@mcm001 is this resolved by SQL changes?

clrozeboom, 2024-04-06:
We've been having this problem lately. On the latest version. No matter what we change, the settings always revert to the last state just before the problem started happening. It's just a few settings and it's manageable but it's annoying. ;)

mcm001, 2024-04-06:
Do you have logs from a boot of Photon for which it happened?

clrozeboom, 2024-04-06:
I'll get some. Is there a way to easily reset the database? I'm fine with Linux commands if need be.

mcm001, 2024-04-06:
This is the most surefire way as of rn:  

```sudo service photonvision stop
sudo rm -rf /opt/photonvision/photonvision_config
sudo service photonvision start```

5010TigerDynasty, 2024-04-06:
[photonvision-journalctl-4-5-24.txt](https://github.com/PhotonVision/photonvision/files/14893764/photonvision-journalctl-4-5-24.txt)
Here's our log file from the last 2 days. Thanks

clrozeboom, 2024-04-06:
FYI, we ended up putting in a spare server which is now working fine. Still have the old one.

mcm001, 2024-04-06:
> No matter what we change, the settings always revert to the last state just before the problem started happening.

Could you elaborate a bit more on what that means? I'm trying to figure out what I should be ctrl-f-ing for in the log you sent

clrozeboom, 2024-04-06:
Ok, for example the 'Top Camera' reverts from using 3-D to using 2-D. That's the most annoying thing that was happening because we wouldn't be able to get pose from it at the beginning of a match.

mcm001, 2024-11-02:
Believe this is OBE after sql and such - reopen if not

closed 2024-11-02 by mcm001