PhotonVision/photonvision #564: Photon Pipeline Index

icemannie, 2022-11-03:
Java getPipelineIndex() returns the pipeline number selected on the UI.

Java setPipelineIndex(n) (n exists) does not cause the pipeline selection on the UI to change.

It does cause the getPipelineIndex() to return the n value. Changing the pipeline on the UI will cause the getPipelineIndex() to return the value of the UI pipeline.

If in Driver Mode on the UI, setting a pipeline number will turn the getDriverMode() to false but won't change the UI switch

mcm001, 2022-11-11:
I suspect all we need is a broadcast full settings call in the handler for setting the pipeline index on a NT change

icemannie, 2022-11-11:
Issue #563 has a similar pron

mcm001, 2022-12-26:
I think making the backend emit a full settings update when the pipeline index is changed over NT should do it

mcm001, 2022-12-26:
```        DataChangeService.getInstance()
                .publishEvent(
                        new OutgoingUIEvent<>(
                                "fullsettings", ConfigManager.getInstance().getConfig().toHashMap()));```

closed 2023-01-04 by gerth2