PhotonVision/photonvision #566: Romi requires Raspberry Pi image

agasser, 2022-11-03:
The Romi installation instructions say, "Follow the process for installing PhotonVision on [“Other Debian-Based Co-Processor Installation”](https://docs.photonvision.org/en/latest/docs/getting-started/installation/install/other-coprocessors.html#other-debian-based-co-processor-installation)." After installing with this script the wrong version of the JAR is installed. The service fails to start because it cannot find the correct native library .so file.

A workaround is to replace the downloaded JAR with the Raspberry Pi version after running the script.

Either the doc should be updated to include this additional step, or the script should install the correct version.

gerth2, 2022-11-03:
I wonder if the instructions are pointing to the wrong jar? The raspi jar
should have that so baked in, and know how to extract it.

agasser, 2022-11-04:
> I wonder if the instructions are pointing to the wrong jar?

That's what it seems like, the script downloads the wrong JAR. The workaround was to get the raspi JAR after running the script. That worked.

mdurrani808, 2022-11-04:
`curl -sk https://api.github.com/repos/photonvision/photonvision/releases/latest |
    grep "browser_download_url.*jar" |
    cut -d : -f 2,3 |
    tr -d '"' |
    wget -qi - -O photonvision.jar
echo "Downloaded latest stable release of PhotonVision."`

Yep, seems like its just grabbing the general .jar file?

mjhealy, 2022-12-16:
+1 to this issue: I'm also seeing this happen when installing on a Romi.

agasser, 2023-01-10:
I don't have a Romi right now to confirm, but I think https://github.com/PhotonVision/photonvision/commit/971b471f92db28b5b008b0dbe9df9880c5ef9ad9 resolves this issue.

mjhealy, 2023-01-11:
If you've got a build that I can try, I'm willing to do so.