PhotonVision/photonvision #570: Feature request: PhotonVision as network tables client for server at localhost

fovea1959, 2022-11-04:
Allow us to configure PhotonVision to be a network tables client for a specific hostname (instead of only a team number). I like to do testing with a simulated robot and PhotonVision running on the same laptop, and it would be very useful to connect PhotonVision to the already-running network tables server at localhost.

mcm001, 2022-11-04:
It wouldn't be a hard change to repurpose the team # entry to do IPv4 or hostname, too, potentially?

fovea1959, 2022-11-04:
I'd be willing to look at the change...

etiennebeaulac, 2023-01-18:
+1, it would be really useful to be able to hook PhotonVision to any NetworkTables server (such as a robot simulation), as said by @gerth2 here https://github.com/PhotonVision/photonvision/issues/260#issuecomment-932771181.

mcm001, 2023-01-18:
If you wanna try implementing plz go ahead! Shouldn't be hard to copy the logic from shuffleboard or something

fovea1959, 2023-01-24:
Halfway there; got the networking and config code fixed, just need to fix the UI. Later this week....

fovea1959, 2023-01-27:
PR is submitted.

Until it is actually pulled, there is a workaround:

Put an entry for a 127.0.0.* in the /etc/hosts ("C:\Windows\System32\drivers\etc\hosts" on Windoze) file for team 9999:

`127.0.0.99 roborio-9999-FRC.lan roborio-9999-FRC.local roborio-9999-FRC.frc-field.local`

If you set your team number in PhotonVision to 9999, it will connect to 127.0.0.99, which should be localhost.

closed 2023-10-28 by srimanachanta