PhotonVision/photonvision #581: New thinclient doesn't work with "npm run serve" locally

mcm001, 2022-11-09:
Stream will show loading forever unless I buildandcopyui and use the server javalin runs at 5800

gerth2, 2022-11-09:
hmm I'll see if I can give this a shot to reproduce. Not instantly obvsiou to me why this would happen, though I assume it's because we're getting stuck somewhere in teh state machine

gerth2, 2022-11-10:
so when I click on the image, it tries to load /thinclient.html, which npm doesn't know about (it's all written outside npm), and that doesn't work. That's expected I think - longer term, for development purposes, we need multiple top-level htmls spit out of one npm build to make that work? 

if I do manually pop the html file open in a new tab and specify a host and port, I see it conneccting to the expected thing:

[image]

thoughts on how this does or doesn't align with what you're seeing?

mcm001, 2022-11-10:
Yeah I think our best bet is to put the thinclient in the public folder of photon-client? I don't love that it somewhat breaks the distinction between UI and thinclient, but it'll keep all our web stuff in one place which is probably a good thing.

closed 2022-12-01 by mdurrani808