PhotonVision/photonvision #583: Cannot open the JAR file for PhotonVision

HarishKaushik15, 2022-11-10:
I am trying to install photonvision and have even followed these steps in this link https://docs.photonvision.org/en/latest/docs/getting-started/installation/install/windows-pc.html. I got to the part where I have to copy and paste a command onto the terminal in order to run PhotonVision and for some reason it is saying that it is unable to access the jarfile. Is there a common reason for this? Any help would be appreciated. Thanks

[image]

mcm001, 2022-11-10:
Hi! The path/to/photonvision needs to be filled in with the actual location of your jar file, which only you know. If you're in the same directory in the command prompt as your jar is in, you can directly run with "java -jar photonvision-whatever.jar".

mdurrani808, 2022-11-11:
an easy way to do this on windows is right click the file, and do copy as path (this is in windows 11, not sure about earlier versions).

closed 2022-12-01 by mdurrani808