PhotonVision/photonvision #592: Could not create an instance of type edu.wpi.first.nativeutils.dependencies.DelegatedDependencySet

ekuznetsov139, 2022-11-11:
I am attempting to build photonvision from source on Raspberry Pi 3 (the OS is Raspbian Buster, though it was manually upgraded from Jessie and some packages may have been missed - that might be somehow causing the problem.)

Following https://docs.photonvision.org/en/latest/docs/other/contributing/photonvision/build-instructions.html, I have installed node.js dependencies. 

When I execute `./gradlew run`  or `./gradlew buildAndCopyUI`, I get an error (the same in either case):

```
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':photon-lib'.
> Exception thrown while executing model rule: Photon(org.gradle.nativeplatform.NativeLibrarySpec) { ... } @ photon-lib/build.gradle line 54, column 9 > all()
   > Could not create an instance of type edu.wpi.first.nativeutils.dependencies.DelegatedDependencySet.
      > java.lang.NullPointerException (no error message)
```

I got a partial stack trace for the exception, but it's not illuminating:
```
Caused by: java.lang.NullPointerException
	at edu.wpi.first.nativeutils.dependencies.AllPlatformsCombinedNativeDependency.resolveNativeDependency(AllPlatformsCombinedNativeDependency.java:48)
	at edu.wpi.first.nativeutils.dependencies.DelegatedDependencySet.resolve(DelegatedDependencySet.java:72)
	at edu.wpi.first.nativeutils.dependencies.DelegatedDependencySet.<init>(DelegatedDependencySet.java:41)
	at edu.wpi.first.nativeutils.dependencies.DelegatedDependencySet_Decorated.<init>(Unknown Source)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at org.gradle.internal.instantiation.generator.AsmBackedClassGenerator$InvokeConstructorStrategy.newInstance(AsmBackedClassGenerator.java:1913)
	at org.gradle.internal.instantiation.generator.AbstractClassGenerator$GeneratedClassImpl$GeneratedConstructorImpl.newInstance(AbstractClassGenerator.java:506)
	at org.gradle.internal.instantiation.generator.DependencyInjectingInstantiator.doCreate(DependencyInjectingInstantiator.java:64)
	... 167 more
```

I may be missing some dependencies not mentioned in bulid docs. In particular, I do *not* have wpilib (it is not clear to me whether it is required, and they don't seem to have any install options for Pi other than a complete new image). But I could get photonvision running on this device out of a prebulit jar.

mcm001, 2022-11-15:
We're seeing the same don Pi platforms. We use wpilib internally for dependency management stuff, probably worth reaching out to them.

mdurrani808, 2022-11-15:
https://www.chiefdelphi.com/t/camera-image-distorted-using-photonvision/400143/12
Seems like this fixed it for them but not for you... did you follow the steps outlined in the docs?

mcm001, 2022-11-15:
Ok per Thadhouse, building on pi is un supported for 2022 wpilib, should be possible when we upgrade to 2023

mcm001, 2023-01-04:
@ekuznetsov139 Any chance you could re-try with beta 8?

srimanachanta, 2023-12-01:
@ekuznetsov139 any updates?

Gold856, 2025-04-14:
Closing as stale.

closed 2025-04-14 by Gold856