PhotonVision/photonvision #595: Client stream view occasionally lost on settings change

amquake, 2022-11-14:
The dashboard UI stream can sometimes "desync" and disappear when settings are changed(the target data is still available). In this video, changing from 2d to 3d while targets are visible causes this. About 30 seconds after the end of the video the stream returned.

https://user-images.githubusercontent.com/7953350/201556856-a26b2891-6577-4305-9af3-d7b4619e7808.mp4

@gerth2 :
> I do still get occasional issues, but actually I think it's more of a whole UI thing (namely, streams getting recreated with new port numbers, but those port numbers not getting broadcast to the front end)

This does not seem to happen when targets are not visible.

@gerth2 :
> i smell a race condition....

mdurrani808, 2022-12-25:
Potentially closed by #649

mcm001, 2022-12-27:
I can still reproduce the desync on latest master. Not sure where the root cause is

amquake, 2023-01-07:
Seems to be fixed by #706 . Would like corroboration before closing

closed 2023-10-09 by amquake