PhotonVision/photonvision #599: Unexpected camera exposure behavior on Arducam OV9281 USB Camera

gerth2, 2022-11-15:
a camera with name `Arducam OV9281 USB Camera` is doing weird exposure things: 

[image]

Assuming cscore will take it, remap full slider range to 0.4-0.5 ?

gerth2, 2022-11-15:
https://discord.com/channels/725836368059826228/725847316707147808/1041921478989459546 Actually, looks like it might be futile. the difference seems to happen as the values `0` or `1` (out of a 0-100 range) are passed into cscore's `setExposureManual()`

gerth2, 2022-11-15:
[image]

asid61, 2022-12-03:
fixes did not work yet. I am willing to send my OV9281 to anyone who wants to debug it...

mdurrani808, 2022-12-25:
@asid61 fix landed in https://github.com/wpilibsuite/allwpilib/commit/149bac55b1a419ca7ed52db9f3f06807ff9a6108
Should be out whenever we cut a new release.

gerth2, 2023-01-04:
Based on discord comments I think it's safe to close this. Can always reopen.

closed 2023-01-04 by gerth2