PhotonVision/photonvision #604: v2023.1.1-beta-6 not seeing 16h5 tags (but does see 36h11 tags).

mperino, 2022-11-17:
[image]
[image]
New install of 2023.1.1-beta-6 sees 36h11 tags just fine, but tried an 8in x 8in tag from:
https://github.com/AprilRobotics/apriltag-imgs/tree/master/tag16h5
specifically tag "07" 

As you can see in the screenshot, it finds and detects the 36h11 tag "42" with no problem..  Regardless of weather there are one or two tags in the FOV it is the same result.  Is there something that I need to do manually to force which family it is detecting?

gerth2, 2022-11-17:
Yes - under the Apriltags tab, there's a selector for which family. Only one family is detected at a time (the tag ID's overlap).

[image]

Give this a shot to see if it fixes up the issue, let us know!

mperino, 2022-11-17:
That fixed it!  Can we make the default tag 16h5 instead of 36h11.
OK to close the issue as fixed.

gerth2, 2022-11-17:
I think switching the default would be ideal. I'll make sure we get an issue in for it - timing wise, I'd expect it should probably be pulled off in whatever release is closest to right before kickoff.

gerth2, 2022-11-17:
Per new title - prior to kickoff, change to the kickoff spec tag family

gerth2, 2022-12-27:
Closed by #652

closed 2022-12-31 by mcm001